tusen-ai/LiDAR_RCNN #2: Collaboration with MMDetection

ZwwWayne, 2021-04-02:
Hi developers of LiDAR R-CNN,

Congrats on the acceptance of the paper!

LiDAR R-CNN achieves new state-of-the-art results through simple yet effective improvement, which is very insightful to the community. We also found that the baseline is based on the implementations in [MMDetection3D](https://github.com/open-mmlab/mmdetection3d).

Therefore, I am coming to ask, as we believe LiDAR R-CNN might have a great impact on the community, would you like to also contribute an implementation of LiDAR R-CNN to  [MMDetection3D](https://github.com/open-mmlab/mmdetection3d)?
If so, maybe we could have a more detailed discussion about that?  MMDetection3D welcomes any kind of contribution. Please feel free to ask if there is anything from the MMDet3D team that could help.

On behalf of the MMDet3D Development Team

BR,

Wenwei

happynear, 2021-04-02:
Hi Wenwei,

Thanks for the assistance when we built the baseline based on MMDetection3D. We are glad to contribute to the open source community. We are refactoring the codes and the codes will be released soon.
 
The algorithm is quite simple. I believe it is not difficult to transplant it to the MMDetection3D. However, the authors of this work are all full-time employees of TuSimple. We may not have enough time to make the codes compatible with MMDetection3D. Maybe your team would have some pioneers to do the work after our codes are released?

Thanks,
Feng Wang

ZwwWayne, 2021-04-02:
OK great.

If so, we will keep update with you. Once your code is released, our team will help to migrate it to MMDet3D. Please feel free to review the PR if you have time or any comments. Your suggestions are valuable to us.

Cheers!

yix081, 2021-04-24:
when will the code be released?

happynear, 2021-05-28:
@ZwwWayne @yix081 Hi all, code has been released.

wangkai602058765, 2021-08-05:
@happynear where can i find the code，thank you！

closed 2021-08-14 by Lzc6996

DongfeiJi, 2021-08-16:
> Hi developers of LiDAR R-CNN,
>
> Congrats on the acceptance of the paper!
>
> LiDAR R-CNN achieves new state-of-the-art results through simple yet effective improvement, which is very insightful to the [...]

Has anyone add lidar-rcnn to mmdet3d?

Lzc6996, 2021-08-16:
@DongfeiJi Not yet.