tusen-ai/LiDAR_RCNN #6: Training from scratch

1349949, 2021-06-02:
Hi guys,
     It's really a nice work! 
     According to the paper and the code，training the pointnet is using the propsals generated by pre-trained detectors(such as pointpillar). 
     So the training of the base detectors and the pointnet is separated.
     I wonder whether you have tried to train the base detectors and the pointnet from scrath. That is, instead of  using pre-trained base detector to generate proposals，training the detector and the pointnet jointly from scrath. 
     If we train the network in this way，the results would be better，worse，or almost the same？

Lzc6996, 2021-06-02:
@1349949 
The advantage of end-to-end training is that the features in the first stage can be jointly trained and reused. However, unlike Faster R-CNN, our model inputs the original point clouds of `[x, y, z]` which is the same as R-CNN. Therefore, there is no need for end-to-end training for LiDAR R-CNN. 
If you try to it end-to-end manner, remember add the feature from 1st stage as inputs to keep the gradiant flow.I didn't try this setting, you can try it anyway.

1349949, 2021-06-02:
Thanks for your replying.

closed 2021-06-02 by 1349949