tusen-ai/LiDAR_RCNN #8: Version of mmdet3d used in the project

haotian-liu, 2021-06-08:
Hi, thank you for your great work. When I tried to generate proposal data of Waymo dataset using the command [here](https://github.com/TuSimple/LiDAR_RCNN/blob/master/tools/data_processer/README.md) for the training split. I modified the code as instructed in the doc, but there was lots of prompts like this:

```
79193 not found.
80014 not found.
79194 not found.
80015 not found.
79195 not found.
80016 not found.
79196 not found.
80017 not found.
```

Is this normal? Or this might due to some code modification of upstream mmdet3d in latest versions. Could you provide the commit hash of the mmdet3d code base you are using? Thanks!

Lzc6996, 2021-06-08:
Hi, @haotian-liu 
Thank you for pointing out the problem.
I have modified our `README`,  added the link to` mmde3d(v0.13.0)`，and corrected the `Line id`.
A little "xxx not found" is by design. You will get `kitti_results_train.bin` about 872M. 
If you still have questions please let me know.

haotian-liu, 2021-06-10:
Thank you for your reply, and the update of the docs is really helpful. I got the code running with the correct change (for 3d-car only). I was facing some other issues in converting the 3class proposals.

1. I can generate proposals by changing waymoD5-3d-3class.py#L137
2. When I run the script for generating the targets `CUDA_VISIBLE_DEVICES='' python3 data_processer.py config/mmdet3d_pp_train_3class.yaml`, there was the following error. This error only occurs for `train_3class` (and it works fine for other 3 configs). My generated `kitti_results_train.bin` for 3class model is 2.2G. Not sure if (1) there are other modifications needed besides the change of the proposal file; (2) there are other information I need to provide for better understand the issue.

```
Traceback (most recent call last): 
File "data_processer.py", line 16, in <module>
    outputs = data_utils.read_bin(cfg.data_path)
File "/disk/code/LiDAR_RCNN/tools/data_processer/data_utils.py", line 56, in read_bin
    objects.ParseFromString(f.read())
google.protobuf.message.DecodeError: Error parsing message
```

PS. One thing that might be helpful for other users who uses different mmdet3d versions (or maybe some polish of the docs, I believe different versions of mmdet3d should all work) is that the `test` and `val` field is identical (L125/L135), so I was actually modifying L125 previously.

Thank you again for your work!

haotian-liu, 2021-06-10:
Problem resolved, thanks!

closed 2021-06-10 by haotian-liu