tusen-ai/LiDAR_RCNN #11: How is the PP model trained

haotian-liu, 2021-06-13:
This model file `checkpoints/hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-car-9fa20624.pth` in the docs cannot be found in mmdet3d official repo (they only have the interval-5 pretrained models). Are the proposals extracted with `interval-1` models: [3d-car](https://github.com/open-mmlab/mmdetection3d/blob/master/configs/pointpillars/hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-car.py) and [3d-3class](https://github.com/open-mmlab/mmdetection3d/blob/master/configs/pointpillars/hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-3class.py)? If I want to reproduce your results, do I need to first train with these two configs? Thanks.

Lzc6996, 2021-06-13:
@haotian-liu 
1. yes we use `interval-1` models
2. Please send me an email for this.  Due to license agreement of the waymo dataset, we can't make it public. Or you can just train models using mmdet3d.

closed 2021-06-13 by haotian-liu

Willforcv, 2021-07-06:
> @haotian-liu
> 
> 1. yes we use `interval-1` models
> 2. Please send me an email for this.  Due to license agreement of the waymo dataset, we can't make it public. Or you can just train models using mmdet3d.

Could you send me the relevant model？ hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-car-9fa20624.pth  and checkpoint_lidar_rcnn_59.pth.tar, thanks

Lzc6996, 2021-07-06:
@Willforcv 
Please send me an email for this

Willforcv, 2021-07-07:
> @Willforcv
> Please send me an email for this

> @Willforcv
> Please send me an email for this

ghw8699536@hotmail.com

Willforcv, 2021-07-07:
> @Willforcv
> Please send me an email for this

ghw8699536@hotmail.com

Willforcv, 2021-07-07:
> @Willforcv
> Please send me an email for this

thanks~

swwwu, 2021-07-16:
@Lzc6996  Hello, author of LiDAR R-CNN.   I'm trying to reproduce your work and I would like to use it after my detection network.  I have the same problem that could not find the trained models.  Could you please send me the models. ( hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-car-9fa20624.pth and checkpoint_lidar_rcnn_59.pth.tar.)   

My email :  shwuuzh@hotmail.com

Thanks a lot.

WangZZJJ, 2021-09-23:
Can you send me the checkpoint_lidar_rcnn_59.pth.tar ?
My email is  wangzijun@ime.ac.cn 
Thanks a lot.

crystal1003, 2021-11-03:
Can you send me the hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-car-9fa20624.pth and checkpoint_lidar_rcnn_59.pth.tar. ?
My email is m18627293699@163.com
Thanks a lot.

Zhangyongtao123, 2021-11-04:
Can you send me the hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-car-9fa20624.pth and checkpoint_lidar_rcnn_59.pth.tar. ?
My email is zhangyongtao6666@gmail.com
Thanks a lot.

Zhangyongtao123, 2021-11-04:
> @Willforcv Please send me an email for this
Hello! Did you get the models(hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-car-9fa20624.pth and checkpoint_lidar_rcnn_59.pth.tar)? If so, please send me a copy of them.
My email is zhangyongtao6666@gmail.com. Thanks a lot.

Zhangyongtao123, 2021-11-04:
> Thanks!

Hello! Did you get the models(hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-car-9fa20624.pth and checkpoint_lidar_rcnn_59.pth.tar)? If so, please send me a copy of them.
My email is zhangyongtao6666@gmail.com. Thanks a lot.

Lzc6996, 2022-01-26:
Thanks for your interest. I have sent the checkpoints to all of you.
I have released the multi-frame training code for LiDAR-RCNN, which leads to AP(L1) 77.8 based on PointPillars' proposals. You can try it anyway. Please feel free to contact me for further discussions!

GatsbyChen, 2022-04-21:
Can you send me the hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-car-9fa20624.pth and checkpoint_lidar_rcnn_59.pth.tar. ?
My email is [18842647783@163.com](mailto:18842647783@163.com)
Thanks a lot.

Lzc6996, 2022-04-24:
@GatsbyChen 
Done!

neisakutou, 2022-04-28:
@Lzc6996 
Appreciate the team a lot for your fantastic work.
I am working on Lidar-RCNN facing the same issue mentioned above.
Could you please send me the pretrained model of 'checkpoints/hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-car-9fa20624.pth'.
Email: ningzuotao@foxmail.com
Hope to hear from you soon.

bruce-almon, 2022-05-03:
@Lzc6996
Thank you for sharing this work.
Can you send me the hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-car-9fa20624.pth and checkpoint_lidar_rcnn_59.pth.tar. ?
My email is skgre1600@gmail.com
Thanks a lot.

fallfo, 2022-07-13:
@Lzc6996 
Thank you for sharing this work.
Could you send me the hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-car-9fa20624.pth and checkpoint_lidar_rcnn_59.pth.tar. ?
My email is [atri2549@163.com](atri2549@163.com)
Thanks a lot.

qingzhouzhen, 2022-09-29:
Thank you for sharing this work.
Can you send me the hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-car-9fa20624.pth and checkpoint_lidar_rcnn_59.pth.tar. ?
My email is 576591769@qq.com
Thanks a lot.