tusen-ai/LiDAR_RCNN #14: Transfered To Nuscenes Dataset，Performance decline

Suodislie, 2021-06-25:
When I transfered it to the CenterpointNet and nuscenes datasets, Then evaluated on nuscense, it didn’t seem to work. I don’t know what went wrong, Looking forward to your suggestions and comments.

Lzc6996, 2021-06-25:
The nuscenes dataset has a multi-frame setting becasue of the 20HZ 32L LiDAR. Do you add the delta time in input points?  @Suodislie

Suodislie, 2021-06-29:
Thank you very much for your reminder, I just added a keyframe point cloud. Have you reproduced this idea in nuscense? In addition, what are the more important steps different from waymo, such as the selection of positive and negative samples, expand_proposal_meter parameters, etc.

Lzc6996, 2021-06-29:
@Suodislie 
I haven't try it in nuscense.

Suodislie, 2021-06-29:
Ok. Ths for your suggestions ！

Lzc6996, 2021-06-29:
@Suodislie If there are some new results in nuscenes please let me know. Feel free to contact me for further discussions!

closed 2021-06-30 by Lzc6996

xxlbigbrother, 2022-01-17:
> 

> hallo！can you share the result of lidar-rcnn in nuscenes?  Performance decline  exactly?

xxlbigbrother, 2022-01-17:
> Ok. Ths for your suggestions ！

hallo！can you share the result of lidar-rcnn in nuscenes? Performance decline exactly?

hutao568, 2022-09-05:
@Lzc6996 hi~can you share your experiment on nuscenes?

Lzc6996, 2022-09-06:
@hutao568 @xxlbigbrother 
Sorry, I didn't conduct a experiment on nuscenes.
For 10 frames point cloud as input，there are so many points not fall in the proposal box. Speed prediction can help us to find these points we need.  But I haven't do it on nuscenes dataset.

qingzhouzhen, 2022-12-30:
Have figure out this problem? I trans this code to nuscnes on pointpillar, which lead to 10 percent decline, @Suodislie

pmj110119, 2023-04-13:
> @hutao568 @xxlbigbrother Sorry, I didn't conduct a experiment on nuscenes. For 10 frames point cloud as input，there are so many points not fall in the proposal box. Speed prediction can help us to find these points we need. But I haven't do it on nuscenes dataset.

Will this algorithm perform worse in scenarios with more detailed categories (such as nuscenes)?

I'm not sure if you have tested on unexpected data from waymo and kitti. Looking forward to your reply!