tusen-ai/LiDAR_RCNN #16: About hyperparameter

zhubinQAQ, 2021-07-07:
hi, guys, this is a good work. I'm curious about the hyperparameters in this work, which hyperparameter is sensitive to dataset? when I apply this methods to my own data, the results is not very good, should I adjust some hyperparameter to get a better performance ?

Lzc6996, 2021-07-07:
@zhubinQAQ 
Can you give a more specific description of your own dataset and "the results is not very good"?
1. Generally, the training set needs enough data to prevent the over-fitting happening in the first stage, just like the KITTI experiment in our paper.  
2. And the distribution of proposals plays an important role which can be controlled by NMS or score threshold.(For too many negative samples, focal loss may be a good choice)
3.You need some visualization to ensure that the preprocessed data is correct.

zhubinQAQ, 2021-07-08:
Thanks for your reply,  I try to experiment on KITTI, and the AP is still low (decrease ~10 bev AP). I find the distribution of proposals(form mmdet3d pp-3class) used for training is bad(Pedestrian,  Cyclist, Car, BackGround ~ 1:1:6:20), did I miss something? Cause I didn't find some filter operation to remove relabeled BackGround class, I try to decrease IoU target but it seems not work. To avoid over-fitting, I test my model on train set, and the results of Pedestrian and Cyclist are bad.

Lzc6996, 2021-07-08:
@zhubinQAQ 
In our paper's Table 3, we show the KITTI results. But we don't recommend do experiments on KITTI datasets becasue  amount of data is too small.  LiDAR R-CNN works by fitting the residuals upon first-stage models, however, without enough data, the model will be very sensitive to parameters, augmentations or tricks.  That's why we focus on Waymo Open Datasets.

Lzc6996, 2021-07-08:
BTW, you can use [score_thresh](https://github.com/TuSimple/LiDAR_RCNN/blob/89c5fefd46563c15f31b8cda76e7ab09e2ddca0d/tools/data_processer/data_utils.py#L100) to control the background.

closed 2021-07-12 by Lzc6996