tusen-ai/LiDAR_RCNN #17: Clarification of the PointNet embedding channels

haotian-liu, 2021-07-12:
Hi, thank you for the great work. I have a bit confusion of the embedding channels in PointNet. In the paper it is said that "We shrink the number of embedding channels to [64, 64, 512] in PointNet to achieve fast inference speed while maintaining accuracy."

However, in the [code](https://github.com/TuSimple/LiDAR_RCNN/blob/master/src/LiDAR_RCNN/models/point_net.py#L16), it seems to be [64, 128, 512]. Could you kindly clarify under which setting are the experiment results in the paper conducted? And how large is the effect of this channel difference (64->128) to the results. Thank you so much!

Lzc6996, 2021-07-13:
@haotian-liu 
That’s a very good question.
It depends on the trade-off of the performance and speed. I will send you a model with [64, 64, 512], you will find it achieves 75.6/66.8(Level1/2) which is the same as the model in [64, 128, 512]. But if you have a more complex task, the former model will not be stable, which means it may become more sensitive to initialization and hyperparameters, because of the too-small model capability.  
So in practice, we prefer the second setting, which is easier to be reproduced, with 1ms speed drop(from 4.4->5.5ms).  If your task is very sensitive to speed, you can just use the first setting, it will be fine for most tasks. If not, I recommend the bigger setting which can save your time from finding a group of suitable initialization and hyperparameters.

haotian-liu, 2021-07-13:
Thank you so much for clarifying!

closed 2021-07-13 by haotian-liu