tusen-ai/LiDAR_RCNN #18: How to test in KITTI dataset？

wangkai602058765, 2021-08-08:
Thank you for your great work！
I am quite interested in your work.I want to know how to run the code in KITTI dataset,it seems that the code you released doesn't support KITTI.Looking forward to your reply.

Lzc6996, 2021-08-08:
@wangkai602058765 
In our paper's Table 3, we show the KITTI results. But we don't recommend do experiments on KITTI datasets becasue amount of data is too small. LiDAR R-CNN works by fitting the residuals upon first-stage models, however, without enough data, the model will be very sensitive to parameters, augmentations or tricks. That's why we focus on Waymo Open Datasets.

wangkai602058765, 2021-08-09:
> @wangkai602058765
> In our paper's Table 3, we show the KITTI results. But we don't recommend do experiments on KITTI datasets becasue amount of data is too small. LiDAR R-CNN works by fitting the [...]

Thank you for your reply！

closed 2021-08-09 by wangkai602058765