tusen-ai/LiDAR_RCNN #19: pc_url_ri2

DongfeiJi, 2021-08-16:
What is pc_url_ri2, is it diffirent from pc_url?
I see pcd = np.vstack([pcd, pcd_ri2]) in load_data func
----------------------------------------------
def get_proposal_dict(data, pc_path):
    outputs_dict = {}
    for o in data.objects:
        output = [
            o.object.box.center_x, o.object.box.center_y, o.object.box.center_z,
            o.object.box.length, o.object.box.width, o.object.box.height,
            o.object.box.heading, o.score, o.object.type
        ]
        key = "{}/{}".format(o.context_name, o.frame_timestamp_micros)
        if key not in outputs_dict:
            outputs_dict[key] = defaultdict(list)
        outputs_dict[key]['pred_lst'].append(output)
        outputs_dict[key]['pc_url'] = '{}/segment-{}_with_camera_labels/{}_1.npz'.format(
                pc_path, o.context_name, o.frame_timestamp_micros)
        outputs_dict[key]['pc_url_ri2'] = '{}/segment-{}_with_camera_labels/{}_2.npz'.format(
                pc_path, o.context_name, o.frame_timestamp_micros)
    return outputs_dict
-------------------------------------------
def load_data(it):
    pcd, pcd_ri2, proposal, gt_box, gt_cls = pkl.loads(it['data'])
    pcd = pcd.astype(np.single)[:, [0, 1, 2]]
    pcd_ri2 = pcd_ri2.astype(np.single)[:, [0, 1, 2]]
    pcd = np.vstack([pcd, pcd_ri2])
    proposal = proposal.astype(np.single)[:7]
    gt_box = gt_box.astype(np.single)[:7]
    return pcd, proposal, gt_box, gt_cls

happynear, 2021-08-16:
It's the second return of a laser. Please refer to https://github.com/waymo-research/waymo-open-dataset/issues/45 for more details.

closed 2021-08-16 by happynear

DongfeiJi, 2021-08-17:
> It's the second return of a laser. Please refer to [waymo-research/waymo-open-dataset#45](https://github.com/waymo-research/waymo-open-dataset/issues/45) for more details.

Thanks very much! BIG ZAN