tusen-ai/LiDAR_RCNN #21: About train one iter data

DongfeiJi, 2021-08-20:
Hi~Sorry to bother you again！
Is that right?
The prediction frames of all frames are extracted at one time and then disrupted globally, which means that when lidar RCNN trains a batch, it contains different boxes of different frames. When the batchsize is 256, the extreme case may contain up to 256 frames, and each frame takes a box.
Below is my idea！
If I train two frames at a time, extract proposals through the frozen one-stage network, and then use lidarcnn for end-to-end training, is it ok？Do u have an idea about how to design the ROI sampler ratio?

happynear, 2021-08-20:
It's OK to train two networks jointly. There are lots of two-stage detection frameworks. You may use their framework and replace ROIAlign with PointPooling, and the RCNN network with Lidar-RCNN to build a two-stage detector.
To step forward, you may query point features from the CNN feature maps and send them as input of LiDAR-RCNN. This may help the network get more context information and potentially increase the performance.

DongfeiJi, 2021-08-20:
> It's OK to train two networks jointly. There are lots of two-stage detection frameworks. You may use their framework and replace ROIAlign with PointPooling, and the RCNN network with Lidar-RCNN to [...]

Is that right?
The prediction frames of all frames are extracted at one time and then disrupted globally, which means that when lidar RCNN trains a batch, it contains different boxes of different frames. When the batchsize is 256, the extreme case may contain up to 256 frames, and each frame takes a box.

happynear, 2021-08-20:
You are right. There is no "frame" in LiDAR-RCNN. Each sample is a bbox.

DongfeiJi, 2021-08-20:
> You are right. There is no "frame" in LiDAR-RCNN. Each sample is a bbox.

Thanks feng ge

closed 2021-08-20 by happynear

DongfeiJi, 2021-08-20:
I just have another question.
what the  Boundary offset means?
It seems like the distance from point(xyz) to proposal whl.
-------
def process_pcd(pcd, proposal, keep_num):
    if pcd.shape[0] != 0:
        # move pcd to proposal's center
        pcd[:, 2] -= proposal[2]
        choice = np.random.choice(pcd.shape[0], keep_num, replace=True)
        point_set = pcd[choice, :3]
        # transform the points to pred box coordinate
        norm_xy = point_set[:, :2] - np.array([proposal[0], proposal[1]
                                               ]).reshape(1, 2)
        point_set[:, :2] = np.dot(norm_xy, rotz(proposal[-1]))
        point_set = np.hstack([
            point_set, -point_set[:, :3] + proposal[[3, 4, 5]] / 2,
            proposal[[3, 4, 5]] / 2 + point_set[:, :3]
        ])
    else:
        point_set = np.full((keep_num, 9), 0)
    point_set = point_set.transpose([1, 0])
    return point_set

happynear, 2021-08-20:
Please look at Figure 3e.

DongfeiJi, 2021-08-20:
> Please look at Figure 3e.

Figure 3e. shows to Front, back, left and right
But code seems add up and down
--------
def process_pcd(pcd, proposal, keep_num):
    if pcd.shape[0] != 0:
        # move pcd to proposal's center
        pcd[:, 2] -= proposal[2]
        choice = np.random.choice(pcd.shape[0], keep_num, replace=True)
        point_set = pcd[choice, :3]
        # transform the points to pred box coordinate
        norm_xy = point_set[:, :2] - np.array([proposal[0], proposal[1]
                                               ]).reshape(1, 2)
        point_set[:, :2] = np.dot(norm_xy, rotz(proposal[-1]))
        point_set = np.hstack([
            point_set, -point_set[:, :3] + proposal[[3, 4, 5]] / 2,
            proposal[[3, 4, 5]] / 2 + point_set[:, :3]
        ])
    else:
        point_set = np.full((keep_num, 9), 0)
    point_set = point_set.transpose([1, 0])
    return point_set