tusen-ai/LiDAR_RCNN #23: expand_proposal_meter

turboxin, 2021-08-24:
Hi there, I noticed that for each 3D proposal its width and length are enlarged to contain more contextual points around it, is it controlled by ''expand_proposal_meter " parameter?

If so, I'm wondering why it's set to 3 meter (so large)? Wouldn't it include nearby points on another car if cars are parked together and confuse the network? And did you guys do abalation studies on this parameter?

Thanks a lot!

Lzc6996, 2021-08-24:
@turboxin 
Thanks for interesting. 
For the first question, yes.
Why is it 3 meter? Actually, we didn't tune this parameter, it's just be set intuitively to include more points.
All points have been transformed into proposals' coordinate, the model can handle such outliers in your description.

turboxin, 2021-08-24:
Thank you for your quick reply~ Closing the issue.

closed 2021-08-24 by turboxin