tusen-ai/LiDAR_RCNN #25: Question about virtual points

mrqrs, 2021-08-29:
Thanks for your nice work!
The paper shows the different pooling method. To add virtual points in the proposal box is well for model result. But I can not find the about code in your project. 
[image]
Can you tell me where the code about 'virtual points', waiting for your replay. 
Thanks.

Lzc6996, 2021-08-30:
@mrqrs 
Thanks for interesting.
In our paper, we propose some methonds to handle the "Size Ambiguity Problem", and we use the ["Boundary offset"](https://github.com/TuSimple/LiDAR_RCNN/blob/89c5fefd46563c15f31b8cda76e7ab09e2ddca0d/src/LiDAR_RCNN/datasets/waymo/data_utils.py#L33-L36) instead of "virtual points" in this public code. Actually, the two methods achieve similar results.

mrqrs, 2021-08-30:
Thanks for your reply.

closed 2021-08-30 by Lzc6996