tusen-ai/LiDAR_RCNN #26: Exceeds maximum protobuf size when preparing training data

xiaoMrzhang, 2021-09-06:
Hi, when I preparing training data following README.md I get some error code and can't generate kitti_results_train.bin data. The error code is as follows:
`Message waymo.open_dataset.Objects exceeds maximum protobuf size of 2GB: 2375722334`

I try to debug this error and find it is caused by the combine operate. The numbers of *.bin files is too big in training data and exceedss 2GB after combined all of them. So I want to know how to deal with this problem, I have tried to split training data in two .bin file but can't concat them in right format.

Thanks

Lzc6996, 2021-09-06:
@xiaoMrzhang 
1. What command or script did you run?
2. What is environment in your server?

xiaoMrzhang, 2021-09-06:
@Lzc6996 Thanks for reply
1. What command or script did you run?
`python tools/test.py configs/pointpillars/hv_pointpillars_secfpn_sbn_2x16_2x_waymo-3d-3class.py work_dirs/hv_pointpillars_secfpn_sbn_2x16_2x_waymoD5-3d-3class_20200831_204144-d1a706b1.pth --out results/waymo-pp-3classes/results_eval.pkl --format-only --options 'pklfile_prefix=results/waymo-pp-3classes_tmp/kitti_results_train' 'submission_prefix=results/waymo-pp-3classes_tmp/kitti_results_train'`
I use single gpu so I replace dist_test.sh as test.py.

2. What is environment in your server?
Ubuntu18 with single 3090 gpu(24G)，128G memory.
By the way, I use mmdet3d v0.17.0 which is different from yours. So the related library such as pytorch, waymo_open_dataset, tensorflow etc are relatively new versions

Lzc6996, 2021-09-07:
@xiaoMrzhang 
You may find some reason for the error you met [here](https://stackoverflow.com/questions/34128872/google-protobuf-maximum-size#:~:text=Protobuf%20has%20a%20hard%20limit,manually%20if%20you%20need%20to.). But actually, I never encounter this problem, and neither did other users. Maybe some recent modifications in mmdet3D introduce this error; perhaps they can give you some advice.

xiaoMrzhang, 2021-09-07:
OK, Thanks for your help @Lzc6996

closed 2021-09-07 by Lzc6996