tusen-ai/LiDAR_RCNN #29: path of data_processer config files

neisakutou, 2022-04-30:
First, thanks a lot for sending me checkpoints previously.

Following the instructions of your work, I would like to double check the following points related to the path of config files.

1. If the **pc_path** and **gt_path** in **config/mmdet3d_pp_train.yaml** and **config/mmdet3d_pp_val.yaml** **share the same path** of the decode points(**pc**) and ground truth(**gt**) generated in the first step.

2. The data_path in mmdet3d_pp_train.yaml is **kitti_results_train.bin**, but  that of mmdet3d_pp_val.yaml(**kitti_results.bin**) was not generated in pp generating.

Thank you so much.

Lzc6996, 2022-05-01:
@neisakutou 
Thanks for your interesting of our works.
for your questions:
1. Yes, the pc and gt are shared.
2. I didn't check the file name in configs carefully. You should keep the "proposals.bin" for train and val from the same one stage model.

neisakutou, 2022-05-02:
Thank you for your reply.

For the above mentioned issue (2) , I used the same pp.bin generated in stage but only switched the mode in train.yaml and val.yaml. 
Consequently, it generated files as following:
target_path
├── train.rec
├── train.idx
├── val.rec
├── val.idx

Thanks!

> @neisakutou Thanks for your interesting of our works. for your questions:
> 
>     1. Yes, the pc and gt are shared.
> 
>     2. I didn't check the file name in configs carefully. You should keep the "proposals.bin" for train and val from the same one stage model.

closed 2022-05-04 by Lzc6996