tusen-ai/LiDAR_RCNN #34: The pretrained model

SSY-1276, 2022-09-08:
Hi, I am very interested in your paper, and I am reproducing it. The pretrained model of pointpillar provided in mmdetection3d does not reach the performance shown in the Table 2 below, so could you please provide the pretrained model of pointpillar in Table 2? Thank you very much!

[image]

Lzc6996, 2022-09-09:
@SSY-1276 
Of cousre, can you give me your e-mail ?

SSY-1276, 2022-09-09:
Thank you for your reply. I'm SSY-1276 in github. My e-mail is ***@***.*** Thank you very much!
Best regards

Lzc6996, 2022-09-09:
@SSY-1276 
Pleasse check your msg, I can get your e-mail.

SSY-1276, 2022-09-09:
My e-mail is yuan_shanshuai@163.com. Thank you

SSY-1276, 2022-09-11:
I reproduce lidar rcnn using your pretrained model. The mAP of vehicle has increased, but the mAP of pedestrain and cyclist each has 10-20 percentage points off. Have you met the problem like this? I have sent an e-mail to you describing this problem in detail.

SSY-1276, 2022-09-13:
[image]
[image]
I reproduce lidar rcnn using your pretrained model, but our results is not same as the results in the paper. Did you meet the question like this?

closed 2022-09-30 by Lzc6996