tusen-ai/LiDAR_RCNN #36: checkpoint shape error

hutao568, 2022-10-17:
hi~ Zhichao Li /Feng Wang/ Naiyan Wang~
我对你们的工作LIDAR RCNN非常感兴趣，但是我在使用您给我的LIDAR RCNN预训练模型checkpoint_lidar_rcnn_59.pth.tar(MD5:6416c502af3cb73f0c39dd0cabdee2cb 时，发现预训练模型的权重是9维，但是你们的输入数据是12维[12维](https://github.com/tusen-ai/LiDAR_RCNN/blob/master/src/LiDAR_RCNN/datasets/waymo/data_utils.py#L38)
您可以提供给我维度可以正确匹配的预训练模型吗

hutao568, 2022-10-17:
[image]

hutao568, 2022-10-17:
[image]
我发现在你们某次commit的时候，维度从9维增加到12维，但是最新的预训练模型依然是9维

Lzc6996, 2022-10-19:
@hutao568 
Hi, sorry for that, we didn't update the pretrain model.
9 -> 12 because mulit-frame and multi return points.
If your don't care the multi-frame setting, you can start from the 9 dim version.

closed 2022-10-19 by hutao568