tusen-ai/LiDAR_RCNN #37: checkpoint shape error

hutao568, 2022-10-18:
hi~ Zhichao Li /Feng Wang/ Naiyan Wang~

I am very interested in your work LIDAR RCNN, but when I use the LIDAR RCNN pretrained model you gave me checkpoint_lidar_rcnn_59.pth.tar(MD5:6416c502af3cb73f0c39dd0cabdee2cb, I found that the weights of the pretrained model are 9 dimensions, but your input data is 12 dimensions.

Can you provide me a pretrained model whose dimensions are correctly matched.

[image]

[image]

I found that in one of your commits, the dimension was increased from 9 to 12 dimensions, but the latest pre-trained model is still 9 dimensions

Lzc6996, 2022-10-19:
@hutao568 
Hi, sorry for that, we didn't update the pretrain model.
9 -> 12 because mulit-frame and multi return points.
If your don't care the multi-frame setting, you can start from the 9 dim version.

hutao568, 2022-10-19:
Can you tell me which branch/commit the pre-trained model of 9dim corresponds to? I'm afraid to find the wrong version

Lzc6996, 2022-10-19:
@hutao568 
I didn't notice this problem. Is your training error decreasing steadily？

hutao568, 2022-10-19:
The training loss is indeed decreasing steadily, but the mAP fluctuation on the validation set will be relatively large, I am not sure if the lr is too large.
[image]
Here are some of my test records, since my baseline (PointPillar) is 0.9% lower than in the paper, I lowered my overall expectations by 0.9%

Lzc6996, 2022-10-19:
@hutao568 
I don't know what happen.
https://pytorch.org/docs/stable/generated/torch.nn.utils.clip_grad_norm_.html   may be helpful

hutao568, 2022-10-19:
Ok, thank you very much for your reply, your work provided a good reference for my research.

closed 2022-10-19 by hutao568

Lzc6996, 2022-10-19:
@hutao568 
One more thing, I have test epoch 57/58/59 the results are similar. In paper, I always report the results from the last epoch which is right setting for validation set.

hutao568, 2022-10-19:
Ok, I will try again to see if the single frame-based case will converge stably, and if increasing the gradient limit in the multi-frame case will alleviate the accuracy fluctuation, thank you.

hutao568, 2022-10-19:
[image]
I got the same results as the paper using the data and pretrained model you provided.
I have an interesting idea, use the LiDAR RCNN model trained by PointPillar's proposal to refine the proposal of a stronger model, such as CenterPoint, will the result of LiDAR RCNN be better than CenterPoint?
If this process does not work, I use CenterPoint's proposal to fine-tune the LiDAR RCNN model. Will the fine-tuned LiDAR RCNN improve the effect of CenterPoint?