tusen-ai/LiDAR_RCNN #39: The num of boxes of matching_gt_bbox is more than that of valid_gt?

QingXIA233, 2022-11-14:
Hello, sorry I came back with another question. Recently, I've been working on using LiDAR R-CNN to refine the results of the CenterPoint-PP model with my own dataset. During data processing for my own dataset, I notice that the results of my CenterPoint-PP model has more bboxes detected than the ground truth ones (false detection case). When performing **get_matching_by_iou** function in LiDAR R-CNN, the obtained matching_gt_bbox has the same number of bboxes as the model predictions instead of the groundtruth data. I'm a bit confused about this process. Now that we are trying to do refinement, shouldn't we remove the falsely detected bboxes in the results and keep to the groundtruth? If so, why the matching bboxes is according to the predictions instead of groundtruth? 

[image]

Maybe I have some misunderstandings here, it would be a great helper if you could give me some hints. Thanks in advance.

Lzc6996, 2022-11-14:
@QingXIA233 
Hi, qing
We use an old-school detection label assignment, assigning a gt box to every positive box. And we judged the positive and negative samples in the [loader](https://github.com/tusen-ai/LiDAR_RCNN/blob/c693828a5ff2d65ab7293c84837df9e3f6a7786b/src/LiDAR_RCNN/datasets/waymo/loader.py#L62).
There is a one-to-one assignment in [DETR,](https://github.com/facebookresearch/detr) which I guess is your want.

QingXIA233, 2022-11-15:
Ok, got it, your method makes sense to me now. Thank you for the quick reply. I'll do it according to your method. Perhaps I'll also perform some experiments with label assignment similar to DETR later.

closed 2022-11-15 by Lzc6996

qingzhouzhen, 2022-11-23:
HI，I am quite confuse about 'gt_cls = 0', I checked the code in 'get_matching_by_iou', it assign the first gt_box to the proposal who match none with all the gt_boxes and coresponding 'gt_cls=0'. Same as this way it did in other places suce as 'relabel_by_iou'. Question is,  'gt_cls=0' is`n vehicle in waymo dataset?(not background class)

Lzc6996, 2022-11-23:
@qingzhouzhen 
HI,  gt_cls = 0 is background.

qingzhouzhen, 2022-11-23:
> @qingzhouzhen HI, gt_cls = 0 is background.

If I transfer dataset to others such as nuscenes, will thie logic work probably？

QingXIA233, 2022-11-23:
@qingzhouzhen Hi, in my understanding, in this repo, they starts the class labels from 1 instead of 0, which means 0 becomes the label that is not considered. My own data' label starts from 0(Car), so I add 1 to each of them before I feed the data to LiDAR R-CNN, then I minus 1 when I run create_results.py after testing the model.