tusen-ai/LiDAR_RCNN #40: What processes in LIDAR-RCNN are specific for waymo dataset?

QingXIA233, 2022-11-21:
Hello, just like the title saying, I wonder what are the specific processes for WOD, which means if I want to use LIDAR R-CNN on my own dataset, I have to do it differently. I already change the **data_processor** and everything I can think of in the **loader** and **creat_results** that are respect to waymo dataset, then I use the refined results to perform evaluation on my own dataset. However, I got NAN on rotation error, and the MAP is pretty low. 
[image]

Therefore, I'm confused about some subtle processes that are performed just for waymo not for other datasets. For example, compute heading residual is necessary for using LiDAR R-CNN? Did you guys use rotation in some sublte ways? （In my dataset, the rotation is according to y axis, while in your code, it's x axis, but the way of computing rotZ is the same, I already changed it.）
[image]

This bug has been driving me crazy, that's why my issue description above is a bit messy, forgive me please. I would be grateful if you could provide me some hints. Thank you a lot. Save this almost desperate kid, please.🥺

QingXIA233, 2022-11-23:
Thank you for the help. Yes, I already changed the files you mentioned above. There are several questions about the code that are confusing me:
1. About the extract_points, is the heading parameter of the pred_bbox_bev important here?
[image]
Because, in WOD, the rotation is around z axis and starting from pos x axis in my understanding, but in my own dataset, the rotation is around z axis and starting form -y axis. So, if heading is used in extrac_points, should I change it according to my own dataset setting? (The heading of my dataset bboxes is shown in the figure below)
[image]

2. In loader.py, when moving gt box to pred center, why use minus here, shouldn't it just be = instead of -= ?
[image]

Lzc6996, 2022-11-23:
@QingXIA233 
1. Yes, 6 from pred_lst is heading.  You should also change the extract_points.
2. this also because the definition of heading. If your dataset is different, please do some visualization to check it is correct.

QingXIA233, 2022-11-24:
Yes, I changed the extrac_points. During the process of moving gt box to pred center, I understand the first line: gt_box[:3] -= proposal[:3}, but why the rotation angle is -propasal heading? In your paper, I see this:
[image]
So, shouldn't the rotation angle be the difference between the gt heading and proposal heading?
BTW, I wonder where I can find the definition of the WOD axes definition (x, y, z and yaw ratation). For example, is the z-axis vertical to the car or the direction of car going?

QingXIA233, 2022-11-24:
Hi, thank you for the warm patience. I just got confused by the transformations in the code:
here, when transfer pcd to pred box coords, you use the rotz matrix of proposal heading
[image]

However, when transfer gt to pred box coords, you use the rotz matrix of  negative proposal heading
[image]

I didn't get the logic here, could you please enlighten me a little? Thanks

Lzc6996, 2022-11-24:
@QingXIA233 
This is the difference between left and right multiplication.

closed 2022-11-28 by Lzc6996

xiuzhizheng, 2022-12-19:
@QingXIA233 
Hi, what performace of lidar rcnn on your own dataset?

QingXIA233, 2023-02-09:
> @QingXIA233 Hi, what performace of lidar rcnn on your own dataset?

Hi, actually, the performance on my own dataset is surprisingly terrible. I have 11 classes, and when I tried to refine the first 4 classes, all of them got around 10 points declination. There is another issue which is about performance declination on nuscenes dataset. The author mentioned about tuning the ratio of the positive/negative. I tuned the IOU threshold and still the refinement is not acheived.

QingXIA233, 2023-02-09:
@xiuzhizheng Did you alo tried on your own dataset? If so, how is the performance.