tusen-ai/LiDAR_RCNN #41: The cls scores are useless on my own dataset

xiuzhizheng, 2022-12-16:
Thanks for your awesome works. When I use Lidar-RCNN on my own dataset, the refine score is useless, Most objects are classified as backgrounds. In addition, the average refined center error is only reduced by 1 cm. I don't know Is this normal?

Lzc6996, 2022-12-17:
@xiuzhizheng 
It looks like you give too much negtive samples in training set.

xiuzhizheng, 2022-12-19:
> @xiuzhizheng It looks like you give too much negtive samples in training set.
Thank you for your apply. The ratio of positive samples is about 0.35. What range should we control this ratio。
One more question, In post-processing (/src/LiDAR_RCNN/utils/eval_utils.py line 78-83), we iterate through each box and assign it a score for each category, But in general, we choose the category with the largest score. Is this a trick?

Lzc6996, 2022-12-19:
@xiuzhizheng 
You ratio seems fine. 
Changing to the maximum score should not greatly affect the results.

xiuzhizheng, 2022-12-19:
@Lzc6996 Have you measured the increase in center point error?

Lzc6996, 2022-12-19:
The Table 8 in our paper may help.

xiuzhizheng, 2022-12-19:
thanks for your help.

closed 2022-12-19 by xiuzhizheng