tusen-ai/LiDAR_RCNN #42: to_pcdet function in loss function

gepu0221, 2023-01-09:
Thank you for your great work! I have a question about loss function.  In this to_pcdet [fuction](https://github.com/tusen-ai/LiDAR_RCNN/blob/8dcf725e91440f480505731ce4af034f521569f8/src/LiDAR_RCNN/ops/iou3d/iou3d_utils.py#L48),  it seems that the dataset format in bbox is [xyz_lidar, l, w, h, rz]. This format looks the same as openpcdet. Why does it need to convert to pcdet format again?

Lzc6996, 2023-01-12:
@gepu0221 
It because the output of models is different from pcdet. The details are shown in [func](https://github.com/tusen-ai/LiDAR_RCNN/blob/8dcf725e91440f480505731ce4af034f521569f8/src/LiDAR_RCNN/ops/iou3d/iou3d_utils.py#L29).
Maybe print it in [line](https://github.com/tusen-ai/LiDAR_RCNN/blob/8dcf725e91440f480505731ce4af034f521569f8/src/LiDAR_RCNN/ops/iou3d/iou3d_utils.py#L36) can help understand.

gepu0221, 2023-01-12:
> @gepu0221 It because the output of models is different from pcdet. The details are shown in [...]
Thanks for your reply! Is the input bbox format of to_pcdet func  [xyz_lidar, l, w, h, rz]? If so, it seems that the format is same as openpcdet. So it doesn't need to convert by to_pcdet func.

Lzc6996, 2023-01-12:
@gepu0221 
Sorry, I can't remember that. But all code in this repo are already test, I think the function should be useful.

closed 2023-01-16 by Lzc6996