tusen-ai/LiDAR_RCNN #44: about aos metric

CocoBoom, 2023-01-13:
thanks for your work！
I apply your work on KITTI dataset.  However, I found that “aos” metric in KITTI datasets performs bad.
In the paper, when the predicted orientation is flipped with 180°， the orientation target is set to be the minimum residual calculated by original orientation and flipped orientation. It doesn't seem to fix the 180° flip orientation error. 
[In this function](https://github.com/tusen-ai/LiDAR_RCNN/blob/master/src/LiDAR_RCNN/datasets/waymo/loader.py#L71)

Lzc6996, 2023-01-16:
@CocoBoom 
Yes, this func is used to fix the ambiguity of the heading loss, but it will not fix the 180 deg flip.
If you want to fix the it, you can add another loss, for example, a classification loss to judge if the heading is flip.

closed 2023-01-30 by Lzc6996

zhangganv5, 2024-04-08:
> thanks for your work！ I apply your work on KITTI dataset. However, I found that “aos” metric in KITTI datasets performs bad. In the paper, when the predicted orientation is flipped with 180°， the [...]

cloud you tell me how to reply it on kitti? THANKS！

CocoBoom, 2024-04-09:
@zhangganv5 I've been on this project for ages, and lost track of the details. Sorry I couldn't be more help.