tusen-ai/LiDAR_RCNN #47: The pretrained model

mobaixin, 2023-06-19:
Hello, I am very interested in your paper, and I am reproducing it. Could you please provide the pretrained model of Table 5. It's about 3D AP results on WOD with three classes trained in one model. My email is 13687162414@163.com. Thank you very much.
[image]

Lzc6996, 2023-06-20:
@mobaixin 
Thanks for your interest. 
I sorry for that we didn't keep the three classes model. I can send you the vehicle class model if you need it.

mobaixin, 2023-06-20:
@Lzc6996 
Thanks for your reply.
That also will be helpful. I need the vehicle class model. Thank you very much again.

mobaixin, 2023-06-21:
I have received your email, thank you very much for your help.

closed 2023-06-21 by mobaixin

xueyingliu, 2024-08-15:
@mobaixin Hi，Could you please send me the pretrained model mentioned above? My email is [1300684011@qq.com](mailto:1300684011@qq.com). Thank you so much~