tusen-ai/LiDAR_RCNN #48: need your pretraind pth model

huang6886, 2023-11-22:
Hello, I am very interested in your paper, and I am now reproducing it. Could you please provide your  pretrained model with car, my email is huang6886@yeah.net, thank you very much

Lzc6996, 2023-11-23:
plz, check your e-mail. @huang6886

huang6886, 2023-11-23:
Thanks a lot , i have received your email.  Another request , do you still have the training/val set（.idx .rec）format( it seems a long time to handle the waymo dataset).

Lzc6996, 2023-11-23:
I don't think the download will be much faster, but anyway, I sent it to you

closed 2023-11-27 by Lzc6996

xueyingliu, 2024-08-20:
@huang6886 Hi，Could you please send me the pretrained model mentioned above? My email is [1300684011@qq.com](mailto:1300684011@qq.com). Thank you so much~

Lzc6996, 2024-08-21:
@xueyingliu Hi, plz check your email, i have send it twice

xueyingliu, 2024-08-23:
@Lzc6996 That’s very kind of you! I’ve received the email now.

wayyeah, 2024-11-11:
@Lzc6996  Hi Could you please send me the pretrained model ? My email is 920258461@qq.com. Thank you so much~

wayyeah, 2024-11-13:
> That’s very kind of you! I’ve received the email now.

@Lzc6996  That’s very kind of you! I’ve received the email now.